antoinejulien/Level-US_GRO400_H21 #19: learn kivy module

franktreppromo65, 2021-02-05:
need to mess arround with kivy interface module so we can be ready when we need to use it

neilroyy, 2021-02-07:
Hey there, how can I be of help? I just studied Kivy a few days back.

antoinejulien, 2021-02-07:
> Hey there, how can I be of help? I just studied Kivy a few days back.

Hi Neil, we are a group of college students and are just starting learning Kivy for our project. We don't know much about it yet, so we don't have questions at the moment. We will keep updating throughout the next months this page. If you want to star the project and se the progress, you are welcome to do so.

antoinejulien, 2021-02-07:
> Hey there, how can I be of help? I just studied Kivy a few days back.

We appreciate your help and if you are willing to help us with futur questions, we would be very grateful.
Thanks again and have a great day,
Level-US team

closed 2021-02-21 by willmarquis